Garr-Garr/OctoPrint-InternalSlicer #3: Import Profile ignores Identifier and Name (and probably Description)

b-morgan, 2024-03-20:
I have a LulzBot TAZ 6 printer configured in PrusaSlicer and executed an Export, Config, and saved the file as config.ini (included in .zip below).

I then clicked `Import Profile...`, browsed to the file on my desktop, and entered an Identifier and Name (I left the Description blank). After clicking `Confirm` the imported config was identified as config and named config. I believe that the Identifier and Name should have come from the `Import Profile...` values.

[image]

[config.zip](https://github.com/Garr-Garr/OctoPrint-InternalSlicer/files/14670902/config.zip)

b-morgan, 2024-03-20:
This install is on an RPi 4B which does not normally have my printer attached. The printer is normally attached to an RPi 3B running OctoPi 0.18.0 and OctoPrint 1.10.0rc3.

Garr-Garr, 2024-10-29:
I'm fairly certain this issue is now fixed in the latest update! I haven't made a new release yet, but you can test it out by downloading the .zip file and uploading it to OctoPrint via the Plugin Manager:

[image]

closed 2024-11-17 by Garr-Garr