Garr-Garr/OctoPrint-InternalSlicer #7: GCODE output is empty

non-bin, 2024-06-30:
This is a fresh install, and the gcode files just contain
```
;Generated from just_in_case.tmp.1719745022467.stl (hash: c984a5cf3c8d6d334d1c4b62464525414c446935)
; BEGIN DISPLAYLAYERPROGRESS SETTINGS
; DisplayLayerProgress_layerIndicatorProcessed = true
; END   DISPLAYLAYERPROGRESS SETTINGS
```
I've tried a few files with the same result.
<details>
  <summary>Here's my config.ini</summary>
  
```
# generated by PrusaSlicer 2.8.0+win64 on 2024-06-30 at 10:32:37 UTC
arc_fitting = disabled
autoemit_temperature_commands = 1
avoid_crossing_curled_overhangs = 0
avoid_crossing_perimeters = 1
avoid_crossing_perimeters_max_detour = 0
bed_custom_model = 
bed_custom_texture = 
bed_shape = 5x0,215x0,215x220,5x220
bed_temperature = 50
before_layer_gcode = ;BEFORE_LAYER_CHANGE\nG92 E0\n;{layer_z}\n\n
between_objects_gcode = 
binary_gcode = 0
bottom_fill_pattern = rectilinear
bottom_solid_layers = 2
bottom_solid_min_thickness = 0
bridge_acceleration = 250
bridge_angle = 0
bridge_fan_speed = 100
bridge_flow_ratio = 0.95
bridge_speed = 25
brim_separation = 0
brim_type = outer_only
brim_width = 0
chamber_minimal_temperature = 0
chamber_temperature = 0
color_change_gcode = M600
colorprint_heights = 
compatible_printers_condition_cummulative = nozzle_diameter[0]==0.4;printer_notes=~/.*PRINTER_VENDOR_CREALITY.*/
complete_objects = 0
cooling = 1
cooling_tube_length = 5
cooling_tube_retraction = 91.5
default_acceleration = 500
default_filament_profile = "Generic PLA @CREALITY"
default_print_profile = "0.16 mm OPTIMAL (0.4 mm nozzle) @CREALITY"
deretract_speed = 40
disable_fan_first_layers = 1
dont_support_bridges = 1
draft_shield = disabled
duplicate_distance = 6
elefant_foot_compensation = 0.1
enable_dynamic_fan_speeds = 0
enable_dynamic_overhang_speeds = 0
end_filament_gcode = "; Filament-specific end gcode \n;END gcode for filament\n"
end_gcode = M400 ; Syncronise planner\n{if max_layer_z < max_print_height}G1 Z{z_offset+min(max_layer_z+2, max_print_height)} F600 ; Move print head up{endif}\nG1 X5 Y{print_bed_max[1]*0.85} F{travel_speed*60} ; present print\nM140 S0 ; turn off heatbed\nM104 S0 ; turn off temperature\nM107 ; turn off fan\nM84 X Y E ; disable motors
external_perimeter_acceleration = 0
external_perimeter_extrusion_width = 0.42
external_perimeter_speed = 25
external_perimeters_first = 0
extra_loading_move = -2
extra_perimeters = 1
extra_perimeters_on_overhangs = 1
extruder_clearance_height = 25
extruder_clearance_radius = 55
extruder_colour = #FCE94F
extruder_offset = 0x0
extrusion_axis = E
[... 267 lines cut ...]
wall_transition_filter_deviation = 25%
wall_transition_length = 100%
wipe = 1
wipe_into_infill = 0
wipe_into_objects = 0
wipe_tower = 0
wipe_tower_acceleration = 0
wipe_tower_bridging = 10
wipe_tower_brim_width = 2
wipe_tower_cone_angle = 0
wipe_tower_extra_flow = 100%
wipe_tower_extra_spacing = 100%
wipe_tower_extruder = 0
wipe_tower_no_sparse_layers = 0
wipe_tower_rotation_angle = 0
wipe_tower_width = 60
wipe_tower_x = 170
wipe_tower_y = 140
wiping_volumes_matrix = 0
wiping_volumes_use_custom_matrix = 0
xy_size_compensation = 0
z_offset = 0
```
  
</details>

jbishop129, 2024-07-02:
Same issue here, with even more sparse .gcode output. I confirmed the source .stl slices fine, produces sane .gcode and prints, directly in PrusaSlicer with the same profile. Below is what the Internal Slicer generates (literally just these 2 lines):
`;Generated from 58mm net pot to 30mm starter plug adapter (3).tmp.1719924499127.stl (hash: 60a0ad417d214c5bd8998733d4cd6e790bfacf6c)
`
Here's my PrusaSlicer .ini file. Printer is a Crealty Ender 3:

> 
arc_fitting = disabled
autoemit_temperature_commands = 1
avoid_crossing_curled_overhangs = 0
avoid_crossing_perimeters = 0
avoid_crossing_perimeters_max_detour = 0
bed_custom_model = 
bed_custom_texture = 
bed_shape = 0x0,235x0,235x235,0x235
bed_temperature = 52
before_layer_gcode = 
between_objects_gcode = 
binary_gcode = 0
bottom_fill_pattern = monotonic
bottom_solid_layers = 3
bottom_solid_min_thickness = 0
bridge_acceleration = 0
bridge_angle = 0
bridge_fan_speed = 100
bridge_flow_ratio = 1
bridge_speed = 60
brim_separation = 0
brim_type = no_brim
brim_width = 0
chamber_minimal_temperature = 0
chamber_temperature = 0
color_change_gcode = M600
colorprint_heights = 
complete_objects = 0
cooling = 1
cooling_tube_length = 5
cooling_tube_retraction = 91.5
default_acceleration = 0
default_filament_profile = 
default_print_profile = 
deretract_speed = 0
disable_fan_first_layers = 3
dont_support_bridges = 1
draft_shield = disabled
duplicate_distance = 6
elefant_foot_compensation = 0
enable_dynamic_fan_speeds = 0
enable_dynamic_overhang_speeds = 0
end_filament_gcode = "; Filament-specific end gcode \n;END gcode for filament\n"
end_gcode = M104 S0 ; turn off temperature\nG28 X0  ; home X axis\nM84     ; disable motors\n
external_perimeter_acceleration = 0
external_perimeter_extrusion_width = 0.45
external_perimeter_speed = 50%
external_perimeters_first = 0
extra_loading_move = -2
extra_perimeters = 1
extra_perimeters_on_overhangs = 0
extruder_clearance_height = 20
extruder_clearance_radius = 20
extruder_colour = ""
extruder_offset = 0x0
extrusion_axis = E
extrusion_multiplier = 1
extrusion_width = 0.45
fan_always_on = 0
fan_below_layer_time = 60
filament_colour = #29B2B2
filament_cooling_final_speed = 3.4
filament_cooling_initial_speed = 2.2
filament_cooling_moves = 4
[... 255 lines cut ...]
variable_layer_height = 1
wall_distribution_count = 1
wall_transition_angle = 10
wall_transition_filter_deviation = 25%
wall_transition_length = 100%
wipe = 0
wipe_into_infill = 0
wipe_into_objects = 0
wipe_tower = 0
wipe_tower_acceleration = 0
wipe_tower_bridging = 10
wipe_tower_brim_width = 2
wipe_tower_cone_angle = 0
wipe_tower_extra_flow = 100%
wipe_tower_extra_spacing = 100%
wipe_tower_extruder = 0
wipe_tower_no_sparse_layers = 0
wipe_tower_rotation_angle = 0
wipe_tower_width = 60
wipe_tower_x = 180
wipe_tower_y = 140
wiping_volumes_matrix = 0
wiping_volumes_use_custom_matrix = 0
xy_size_compensation = 0
z_offset = 0

Edevige, 2024-10-21:
Post the log of the plugin, i had the same issue and it was just a missing library

Garr-Garr, 2024-10-29:
@Edevige Any idea which dependency was missing? I'd be happy to help diagnose what's going on and try to further automate the installation process. 

@jbishop129 @non-bin Could you try to slice a new file again and then download the OctoPrint log files? 
 - OctoPrint Settings Menu -> System Info -> Download Systeminfo Bundle

[image]

Edevige, 2024-10-29:
@Garr-Garr it was **libgtk-3**, unfortunately i can't post my log to help you further.
Thank you for this project

n2qcn, 2024-11-10:
yup, an `apt install libgtk3.0-cil` fixed it. Thanks!

Garr-Garr, 2024-11-11:
> yup, an `apt install libgtk3.0-cil` fixed it. Thanks!

Thank you for the update! Do you know which OctoPi version you're running? I think some newer versions must have that preinstalled since I haven't ran into this issue yet, but the systeminfo bundle would give me a better understanding:
 
- OctoPrint Settings Menu -> System Info -> Download Systeminfo Bundle

n2qcn, 2024-11-11:
> env.plugins.pi_support.model: Raspberry Pi 4 Model B Rev 1.5
env.plugins.pi_support.octopi_camera_stack: webcamd
env.plugins.pi_support.octopi_version: 1.0.0
env.plugins.pi_support.octopiuptodate_build: 1.0.0-1.10.3-20241105092907
env.plugins.pi_support.octopiuptodate_build_short: 2024.11.05.092907
env.plugins.pi_support.throttle_check_enabled: True
env.plugins.pi_support.throttle_check_functional: True
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 22.3
env.python.version: 3.9.2
env.python.virtualenv: True
octoprint.last_safe_mode.date: 2024-09-06T19:20:34Z
octoprint.last_safe_mode.reason: settings
octoprint.safe_mode: False
octoprint.version: 1.10.3
systeminfo.generated: 2024-09-11T19:56:05Z
systeminfo.generator: zipapi

closed 2025-01-01 by Garr-Garr