Garr-Garr/OctoPrint-InternalSlicer #8: Update to newer PrusaSlicer versions (2.8.1?)

krisrok, 2024-11-09:
Hi, I saw you already tried to support newer versions on OctoPi 0.18.0 but I wonder if anything's changed now there's 1.0.0 with bullseye inside?

I also found this article: https://github.com/prusa3d/PrusaSlicer/issues/13237
They managed to get the UI running, but not the slicing part (via UI I guess). So maybe slicing via commandline would work. I just added a comment asking about the OS version.

Maybe you can already see potential issues with this? E.g. other integration steps you need to take for the plugin? Otherwise, I'd upgrade and just give it a go.

Garr-Garr, 2025-01-01:
What PrusaSlicer v2.8.1+ feature are you wanting this plugin to implement? I've been focusing on stability rather than chasing the highest version number. Most new features can't really be implemented in the web GUI and prusa is no longer focusing on supporting appimages

the article you found is just a weird way to render the program's GUI on a CM4 , and wouldn't be compatible within an OctoPrint web page instance...?

closed 2025-01-01 by Garr-Garr

krisrok, 2025-01-02:
No feature in particular. Mostly because of profile parity between workstation and pi.

I just linked the article because they managed to get 2.8.1 running. Not because of the UI, just in general. Maybe commandline slicing would work.

No worries if it‘s too much of a hassle to implement though.