Garr-Garr/OctoPrint-InternalSlicer #9: Slicing Disabled - Slicing Profile dropdown empty.

b-morgan, 2024-11-17:
I installed version 1.0.3 on OctoPrint 1.10.3 (OctoPi 1.0.0cam), installed the PrusaSlicer, exported the config.ini file from PrusaSlicer 2.8.1, and imported the profile on my RPi 4B. The profile did get named correctly (issue #3) but after uploading a small 5x5x5mm cube .stl file, I got the Slicing Disabled... message and the dropdown for the Slicing Profile is empty. I hope I have attached all the relevant files but let me know if you need anything else. Note: The LulzBot TAZ 6 is not currently connected to the RPi 4B. It is attached to an RPi 3B.

[config.zip](https://github.com/user-attachments/files/17791133/config.zip)

[image]

[image]

[octoprint-systeminfo-20241117083647.zip](https://github.com/user-attachments/files/17791121/octoprint-systeminfo-20241117083647.zip)

SteveGSY, 2024-12-01:
I have the same issue thought I had done the configs wrong.
Hope this gets fixed looks to be a useful resource.

koehdaniel, 2025-01-03:
I just ran into the same problem. Did anyone find a solution?
@Garr-Garr do you know any quick-fix / workaround?

Warm regards, 
Daniel

amaisano, 2025-01-10:
I have the Prusa MK3 and I found I needed to rename the details, description, id, etc, in the upload dialog to not include periods or special characters to import he profile. However, I'm stuck at the same problem as others. The profile is there in the plugin settings, but after many saves and reboots, i cannot select it in the Internal Slicer and slicing is disabled...

amaisano, 2025-01-10:
AH I got it working. I went into the Profiles tab in the settings and added a test profile. This is a different interface from the Internal Slicer profile upload tool. I called it TEST and then after saving, the real one "MK3..." was automatically selected on the internal slicer tab:

[image]

amaisano, 2025-01-10:
Nevermind... it helps unlock the Slice It! Button, but it's generating zero gcode files - just blanks.

bondiblueos9, 2025-01-14:
I am seeing the same behavior. Empty drop down to select profile.

andybrinkley, 2025-08-04:
I found an answer to this. If you go into the settings (expand it if you need to)

[image]
Then click 'set as default', then the dropdown will work.
One more thing though - on mine, I checked the logs, and found that I was also missing some library (libgtk-3.so.0 I think)
So I had to install that from the command line, which made it all work.

andybrinkley, 2025-08-04:
This is where I found the settings:

[image]

amaisano, 2025-08-04:
I do not have the "Internal Slicer" plugin, so that explains it, haha.

cletusw, 2026-02-10:
+1 I had to do the default slicer thing from #issuecomment-3150737753 *and* manually `sudo apt install libgtk-3-0 fuse` to get it to work & stop outputting an empty gcode file